MiSTle-Dev/C64Nano #2: GPIO normalization

misterjbam, 2023-10-15:
Good morning

Would it be possible to standardize the GPIO port?
For the joystick port of this core, is it possible to replace it to use the same location and controller as the NESTANG, PACMAN, GALAGA, ... cores?
And for the keyboard, would it be possible to standardize it with the core developer Atari ST?
If all cores are standardized at the GPIO level, it will be simpler for users.

Thanks to all the developers

misterjbam, 2023-10-15:
the developer of the core atai ST wants to do without GPIO and run everything via USB (keyboard, mouse, controller)
Do you think it would be possible to implement on the C64 core?

vossstef, 2023-10-15:
Hi, if i am not totally mistaken there are already serveral projects that are using FPGA IO pins with low speed USB 1.1 devices (keyboard, mouse, joystick) even without the dedicated USB tranceivers like USB3317 or MAX3421E. Interfacing needs would collapse just into USB connector, resistors and maybe fpga input protection. See e.g. github.com/nand2mario/usb_hid_host/tree/main
At the moment i am concentrating more on burning tasks like proper HDMI output and porting memory from bram to dram rather thoughts of moving away from good old PS2 KB and beloved legendary competition pro digital joystick... but time will come also for that. However if such GPIO 'normalization' would become available would be of course interested to utilize it.

vossstef, 2023-10-18:
> As an intermediate step I am currently programming a M0S Dock (also made by Sipeed) to act as a USB/PS2 bridge to give the MiSTeryNano access to a USB keyboard and a USB mouse. The same solution should work for your C64 core as well.
> 
> In the long term I am planning to do this with the BL616 already present on the Tang Nano 20k.

I see and thanks for notification.
Guess we would we need something like a 4 port USB2.0 hub to connect keyboard, mouse and two gamepads or alternately  two joysticks isn't it ? Looks to me like excellent scalable and flexible soluation rather than growing amount of diff pair needs per HMI input device at FPGA.

harbaum, 2023-10-19:
These mini USB hubs fit nicely: https://a.aliexpress.com/_EIidgjH

Also my primary interest for this are small cheap wireless keyboard and touchpad combos to give the Atari ST a keyboard and a mouse. These integrate both devices and thus only one small USB dongle is needed. Together with the necessary USB-C/USB-A adapter this is still very small.

closed 2023-11-04 by vossstef

harbaum, 2023-11-08:
Did you try one of those 5 inch 800*480 LCDs they sell for the TN20k? These cope nicely with 50hz and 31.5kHz signals and can e.g. display all the Atari ST video modes.

I am sure they'd work with your core as well. But as the lcd uses many of the IOs only few are left for peripherals and allowing for at least spi or PS2 together with the lcd would probably require yet another relocation of pins.

Btw: these devices: https://www.notjust4nerds.de/Riitek-Rii-i8-Mini-DE-Funk-USB-Dongle-Tastatur-QWERTZ-Maus-Kombo-beleuchtet-kabellos

work nicely with the Tang Nano. They work as mouse and keyboard and I consider mapping those top left buttons to joystick.

vossstef, 2023-11-09:
I did make use of the Sipeed SH500Q01Z LCD-TFT with CoG TCON Ilitek ILI6122  https://api.dl.sipeed.com/shareURL/Accessories/LCD in the VC20 project https://github.com/vossstef/tang_nano_9k_vic20_lcd  and it is working rather fine.  Just a bit scared to break the flexfoil when plugging and unplugg it from TN too often.  This Sipeed LCD does not have any touch sensor. TN20k seem to be rather limited in available IO when it comes to parallel LCD (or old VGA) interfacing. Maybe the Tang Primer 20k + full blown Dock would be a better choice as far more IO's available.