MiSTle-Dev/C64Nano #4: black screen

misterjbam, 2023-10-18:
Since the HDMI fixes made yesterday, I can no longer display an image.
I tested on different monitor.

vossstef, 2023-10-19:
looks like by adjusting the main PLL for improved HDMI compatibility a stability issue popped up. 
will fix that in the neare future and revered temporary back to 40k bram as back-up.

misterjbam, 2023-10-20:
thanks, it works again

vossstef, 2023-10-21:
Hi misterjbam,
would you please be so kind and check on your setup this dram development branch ?
Mainly i simply adjusted DRAM timing....

older branch was working on one of my TN20k but on my 2nd spare part it also showed same black screen phaenomen.

misterjbam, 2023-10-21:
I just tested dev_sdram and it's ok, I have the HDMI working

vossstef, 2023-10-22:
thanks and will merge now

closed 2023-10-22 by vossstef