MiSTle-Dev/C64Nano #12: Use flash ROM for ROM and cartridges

harbaum, 2023-11-08:
The Tang Nano 20k comes with a 64MBit/8Mbytes SPI flash. The first 700+ kBytes are used to store the FPGA bitstream. But the remaining 7MB are unused and can be accessed by the FPGA at runtime.

This can either be used directly and with DSPI mode at 100MHz a random read speed of <250ns can be achieved. This should be sufficient for the C64. Especially multiple ROM carts could be stored there. The flash may also be used to initialize some of the SDRAM at startup such that the SDRAM can also be used to store Rom data but with a shorter access time.

vossstef, 2023-11-08:
thanks for the very good idea and will remember when i managed a bigger refactoring to be more compatible to Mister and the more urgent missing OSD function.

harbaum, 2023-11-11:
The MISTer osd is somewhat outdated. I am currently implementing a slightly more modern osd for misterynano. Perhaps you'd like to cooperate ....

harbaum, 2023-11-11:
First tests with the new OSD:

https://m.youtube.com/shorts/zsHYcolqtpc

Next will be the floppy disk image selector for that. The idea is that the BL616 deals with the SD cards file system and let's you select a disk image file. And whenever the core wants to read a sector from floppy then the BL616 translates the original sector address into a sector address within the SD cards own file system. So the FPGA would still read sectors from the floppy disk image by itself, but the sector numbers would be translated by the BL616 which has plenty of ram and plenty of CPU power to deal with the SD cards file system.

vossstef, 2023-11-20:
> The MISTer osd is somewhat outdated. I am currently implementing a slightly more modern osd for misterynano. Perhaps you'd like to cooperate ...
> 
> Is there a way to send a PM through GitHub?

github PM had been some while ago removed. I send today a PM.

closed 2024-02-12 by vossstef