MiSTle-Dev/C64Nano #18: Broken audio output

prcoder-1, 2023-12-27:
Commit  **63c4e59a18274a8126092167541f447758e148f7** (audio volume fix) breaks sound output.
Hissing over-amplified sound. And the output is only the left channel for some reason.

vossstef, 2023-12-28:
@prcoder-1 would you pls. try this correction and let me know ? 
Control Audio Volume via OSD (F12 Key) Thx !

prcoder-1, 2023-12-28:
> @prcoder-1 would you pls. try this correction and let me know ? Control Audio Volume via OSD (F12 Key) Thx !
I tried adjusting the volume through the OSD menu (F12). Only the "Off" option works. The rest (33%,66%,100%) work exactly the same and output over-amplified, distorted sound.

vossstef, 2023-12-28:
strange... Added even rather new HMDI core enhancements and Audio processing identical to Mister main branch. In my opinion there should be a difference. There was a clearly noticeable difference on my Monitor. 
I added for your convenience the .fs file too.

https://github.com/vossstef/tang_nano_20k_c64/blob/dev_audiofix/impl/pnr/tang_nano_20k_c64.fs

May you try this ready made synthesized .fs file and let me know ? Sorry for inconvenience. Thx

prcoder-1, 2023-12-28:
Your .fs file work fine. Sounds right. But volume control doesn't work. Only 'Off' for no sound and other value for sound on.

After these changes, the sound worked fine for me in the version I synthesized:

`diff --git a/src/tang_nano_20k_c64_top.vhd b/src/tang_nano_20k_c64_top.vhd
index edacadd..57bcdae 100644
--- a/src/tang_nano_20k_c64_top.vhd
+++ b/src/tang_nano_20k_c64_top.vhd
@@ -397,8 +397,8 @@ port map(
       g_in      => std_logic_vector(g(7 downto 4)),
       b_in      => std_logic_vector(b(7 downto 4)),
 
-      audio_l => '0' & audio_data_l(17 downto 11) & audio_data_l(17 downto 10),
-      audio_r => '0' & audio_data_l(17 downto 11) & audio_data_l(17 downto 10),
+      audio_l => audio_data_l(17 downto 2),
+      audio_r => audio_data_r(17 downto 2),
       enabled => osd_status,
 
       mcu_start => mcu_start,
`

I don't understand how this logic works and why only the left channel (audio_data_l() is used in both cases (audio_l and audio_r ):
`'0' & audio_data_l(17 downto 11) & audio_data_l(17 downto 10),'

prcoder-1, 2023-12-28:
Branch **dev_audiofix** synthesized and works well. But the sound volume is not adjustable, it can only be turned off.

closed 2023-12-29 by vossstef