MiSTle-Dev/C64Nano #35: Feature request: Datel MIDI cart

harbaum, 2024-02-07:
I have a carrier board for the Tang Nano with MIDI IN and OUT. Unfortunately this collides with the Dual shock connector of the C64 core.

I still would like to see an option to use the MIDI ports in the C64 core. This has been done for the MiST C64 core as it seems:

https://github.com/mist-devel/c64/issues/1

The use case? I think it would be quite funny to couple two Tang Nano via MIDI, run Atari ST + Cubase on one of them and C64 + Cynthcart on the other one. The C64 with it's SID would then be used by the Atari ST as a MIDI synthesizer.

Admittedly a rather odd setup and probably not the most common use case ... but nevertheless tempting ...

I might give that a try myself. But maybe you actually derive your code from the MiST core.

vossstef, 2024-02-08:
will add the already existing MIDI function so no issue.
Need to fix someother topics first.

harbaum, 2024-02-08:
No need to hurry ... This is really just for a little personal entertainment ...

Thanks!

vossstef, 2024-02-09:
I added midi to following branch:
https://github.com/vossstef/tang_nano_20k_c64/tree/c1541dos_2_flash

Can't test as i don't have the interface pcba and knowledge of Synthesizer's.
Be able to play a Korg M1 would be nice but reality...  

The design still has a weird behavior in locking the video signals that i have to fix

vossstef, 2024-02-12:
released

closed 2024-02-12 by vossstef