MiSTle-Dev/C64Nano #40: Black screen - no blue prompt and no loading

camplusplus, 2024-02-17:
I flashed it with Linux and Windows afterwards but no blue commodore prompt displays on boot as in your screenshots.

Also cannot load any disk image.

Only f12 menu works.

vossstef, 2024-02-17:
The fpga design can only live together with a M0 Dock with programmed misterynano firmware for the bl616 µC.
To me it looks that the design is kept in reset by missing bl616 response on SPI interface.
The large multi color LED on the TN need to be illuminated in green.
In addition need one of the other LEDs need to be illuminated in red indicating that the m0S dock detected.
pls. checking wiring and correctly programmed µC fw. 
0.1.7 requires the fw that i adhoc compiled and attached in the release package.

camplusplus, 2024-02-17:
Large multi color LED on the TN is Green
One smaller led is amber on TN.

3 red are on M0S dock. 
I will check connection but keyboard works

camplusplus, 2024-02-17:
On boot i have next leds counting from 0 from usb: 

0: quickly flashes
3: solid light M0S Dock detect

camplusplus, 2024-02-17:
I will try with older pairs of firmware

camplusplus, 2024-02-17:
Also same luck with v0.1.5
I flashed both M0S and TN with files from release.

Menu works fine, file browser working but black screen.

harbaum, 2024-02-17:
Did you start with a fresh SD card? Not that you have some old saved settings on card which prevent the C64 from booting.

camplusplus, 2024-02-17:
Button mechanically works also tried with fresh SD card Formatted in fat16.

I'm suspecting IRQ fin form M0S

camplusplus, 2024-02-17:
Formatted SD card to FAT16

Reset button mechanically clicks.

IRQ pin is 3.24 V

camplusplus, 2024-02-17:
Same issue with Mistery Tang, which was working before.
I'm trying to erase flash completely

harbaum, 2024-02-17:
MiSTeryNano would even boot without any M0S connected and no SD card inserted at all.

But you need a TOS ROM in flash. Otherwise it won't boot.

If it worked before and now doesn't I'd check if the reset button is working even if it clicks. There's not much more that may prevent both cores from booting while e.g. the menu works.

camplusplus, 2024-02-17:
I added TOS-es for MiSTeryNano.
Reset button working in MiSTeryNano.

Same situation in MiSTeryNano: Menu working but black screen.

harbaum, 2024-02-17:
How do you know that reset is working if you don't see anything?

camplusplus, 2024-02-17:
Monitor reconnects

harbaum, 2024-02-17:
What vertical frequency does your monitor report in MiSTeryNano? If it's 50Hz then the machine actually executed some part of the TOS operating system.

camplusplus, 2024-02-17:
it's was 50Hz

harbaum, 2024-02-17:
I am running out of ideas. My current guess would be that the embedded SDRAM of the FPGA has failed ...

camplusplus, 2024-02-17:
I would say also that I have defective board

closed 2024-02-17 by camplusplus