MiSTle-Dev/C64Nano #52: [Question] SID and VICII from MiSTer?

vanfanel, 2024-05-05:
Hi there, @vossstef 

I just discovered my Tang Nano 20k can do C64 thanks to this!
Now, are the SID, VIC and CPU directly from the MiSTer project? Both are pretty much " perfect" on the MiSTer by now, and I would like to know what to expect with this :)

Thanks!

vossstef, 2024-05-05:
SID, c1541 and MIDI is from MiST, VIC, CIA and CPU from MiSTer. I enhanced e.g. c1541 with Dolphin DOS support and did entire new top level integration based on MiSTer SystemVerilog sources with mandatory changes in PLL and HID integration in VHDL. Some modules are entirely now like the sdcard loader and the block memory. 
In my mind is it working rather smooth but please try it out and report if something is not working as expected.

vanfanel, 2024-05-05:
I will test as soon as I can (I don't have the tang nano 20k here) and report back!

About the SID, isn't the MiSTer SID more precise? Or maybe I am mistaken?

vossstef, 2024-05-05:
Please take care that you order the Sipeed M0S Dock as well https://wiki.sipeed.com/hardware/en/maixzero/m0s/m0s.html
If i recall right i had problem to compile the MiSTer SID a while ago therefore i use the MiST one.

vossstef, 2024-05-10:
@vanfanel any news related to function of the C64 core ? 
Is it working ? Can the issue be closed ? thx

vanfanel, 2024-05-10:
@vossstef Thanks for caring :)
It was more a question than an issue, really.
I got the core to work partially because I still need to update the M0S firmware.
But this can be closed indeed. Thanks!

closed 2024-05-10 by vanfanel