MiSTle-Dev/C64Nano #53: Trying to flash BL616 on Linux

vanfanel, 2024-05-09:
Hi,

I have started to try this C64 on my tang nano 20k + dock, which has M0S Dock onboard:

[image]

I have no problem with flashing the core with this command:
openFPGALoader -f tang_nano_20k_c64.fs
It boots!

However, I am missing many options on the OSD, like CRT loading etc.

Looking at the documentation here:  https://github.com/vossstef/tang_nano_20k_c64/blob/main/INSTALLATION_WINDOWS.md
I can see the BL616 on the M0S Dock has to be updated too.
For that, you point to a program called BuffaloLabDevCube which seems to be Windows software. I don't have a Windows machine, and will never use that OS.
Is there a way to do that on GNU/Linux?

Thanks!

harbaum, 2024-05-09:
https://github.com/harbaum/MiSTeryNano/blob/main/INSTALLATION_LINUX.md

But as I said in the other thread: Please ask the vendor of your board for help first. He has initially flashed it and he knows how to update it. He's making the money and he's the one to get you started and tell you how to use the board. We have never seen nor used your board.

vanfanel, 2024-05-09:
> https://github.com/harbaum/MiSTeryNano/blob/main/INSTALLATION_LINUX.md
>
> But as I said in the other thread: Please ask the vendor of your board for help first. He has initially flashed it and he [...]

That's what I needed. Thanks, Till. Sorry to bother you.

vossstef, 2024-05-09:
Be aware that at this moment in time C64 and VIC20 core enhanced BL616 firmware has be loaded including many new CBM core features like the lately added loaders. I think that the latest FW i complied and added under releases will support all three cores.

harbaum, 2024-05-09:
Indeed. Everything will hopefully get a little more stable over the time. But currently there may be versions that may not support all cores fully at the same time.