MiSTle-Dev/C64Nano #56: Video and audio problems

vanfanel, 2024-05-17:
Hi @vossstef! 

I could finally test the latest core version on the Tang Nano 20k after flashing the BL616 with the latest firmware, I'm having A LOT of fun with it!

However, I've noticed some video & audio problems:
-If `Settings->Screen` is set to `Wide`, there is no HDMI audio. `Wide` is needed for correct aspect ratio on modern 16:9 monitors.
-If `Settings->Screen` is set to `Wide`, screen is not centered: it's displaced towards the left side of the screen.
-SID audio doesn't match the MiSTer core. For example, try Tim Follin's Ghouls'N Ghosts tittle screen music: in both 6581 and 8580 modes, with or without DIGIFIX, it doesn't sound correct (it sounds correct on the MiSTer core).
This is how it sounds on real hardware and on the C64 MiSTer core:  https://www.youtube.com/watch?v=wz36JWHTk-A
Note this music is perhaps the definitive C64 audio test: only the C64 MiSTer core and real C64 passes this hard test :)

vossstef, 2024-05-17:
> Hi @vossstef! 
> 
> I could finally test the latest core version on the Tang Nano 20k after flashing the BL616 with the latest firmware, I'm having A LOT of fun with it!
> 
> However, I've noticed some video & audio problems:
> 
> -If `Settings->Screen` is set to `Wide`, there is no HDMI audio. `Wide` is needed for correct aspect ratio on modern 16:9 monitors.
> 
> -If `Settings->Screen` is set to `Wide`, screen is not centered: it's displaced towards the left side of the screen.
> 
> -SID audio doesn't match the real C64 or the MiSTer core.
> For example, try Tim Follin's Ghouls'N Ghosts tittle screen music: in both 6581 and 8580 modes, with or without DIGIFIX, it doesn't sound correct (it sounds correct on the MiSTer core).
> This is how it sounds on real hardware and on the C64 MiSTer core:  https://www.youtube.com/watch?v=wz36JWHTk-A
> Note this music is perhaps the definitive C64 audio test: only the C64 MiSTer core and real C64 passes this hard test :)
> Are filters supposed to be implemented? Maybe it's simply filters being missing at this point?

Hi,
so far i didn't focus too mich on 16:9 but will have over time a Look at it.
Concerning the Mister SID i hat a lot of Compiler  Problems and therefore picked the Mega65 SID variant. Didn't know those mentioned game as reference but checked some rather New Demos that sounded about correct to my ears. Also the dual sid in Primer25k looked to me correct. Good news is that i found a lot of linting errors in the Mister sid and got it fixed and running with the Gowin IDE. Will check with that your game for Sound  differences and maybe swap.

vanfanel, 2024-05-17:
@vossstef The fact that you got the MiSTer SID working on the Gowin IDE is fantastic!

If you test Ghouls'n Ghosts, please use original 6581 SID, since the game was designed with that SID variant in mind (all Tim Follin soundtracks are meant to be played on a 6581 since that's what Tim had at home back in the day, apparently).

vossstef, 2024-05-22:
@vanfanel may i ask you a question...
Is in a MISTer FPGA core the SID Filter Option valid for 6561 or both 6561 and 8580 ?

vanfanel, 2024-05-23:
@vossstef Good question.
In MiSTEr C64 core, SID Filter option is only available for 6581, but not for 8580.
As soon as 8580 SID is selected, SID Filter option becomes greyed out.
I have recorded a video so you can see it better.

https://github.com/vossstef/tang_nano_20k_c64/assets/837585/8826f7b2-fe06-4218-8849-284b405ab3e9

vossstef, 2024-05-23:
@vanfanel 
i created a new branch for testing and dropped the tang_nano_20k_c64.fs file and the needed misterynano_fw_bl616.bin
You will find all here: https://github.com/vossstef/tang_nano_20k_c64/tree/55-tape-sound-and-userport-uart-option

please check Cartridge and SID and give feedback. Big Thanks for help !

Changes:
SID from MISTer (incl. Filter Options) 
Cartridge fixed (MISTer)
RS232 UART
Tape Sound
I was not able to detect sound changes while playing around with the SID Filters. Maybe you have more luck.

vanfanel, 2024-05-24:
I am so grateful for this! Perfect SID means a lot to me, this chip had a truly magical sound...

closed 2024-05-24 by vossstef

vossstef, 2024-06-11:
Hi @vanfanel , i created a proposal for a new release with new features:
New features:
- 2nd SID option (Stereo) also for TN20k
- SID Filter options : Follin, Galway, Average, Strong, Extreme
not yet working: Turbo-232 serial Interface preparations

Bugfixes:
- digital Joystick DB9 fixed
- DualShock Paddle Emulation fixed
pls. flash new misterynano fw and new fpga bitstream in branch
https://github.com/vossstef/tang_nano_20k_c64/tree/63-porting-of-latest-redip-sid-pipelining-mode-for-tn20k-dual-sid-usecase
pls report any issues and many thanks for checking . thx !

vanfanel, 2024-06-11:
@vossstef I have flashed the latest .fs and .bin, but there's some problem with the bl616 fw bin.
I am flashing exactly this file:
https://github.com/vossstef/tang_nano_20k_c64/raw/63-porting-of-latest-redip-sid-pipelining-mode-for-tn20k-dual-sid-usecase/misterynano_fw_bl616.bin
...But it's missing all SID options. Maybe you uploaded the wrong file? An old version maybe?

vossstef, 2024-06-12:
@vanfanel just added again the firmware and fw.ini
Would be glad if you could check SID sound. 
SID Filter settings like 'exteme' i checked with Ghosts_Goblins and there is a difference now. Not sure was my previous MISTer port behaving like that. Do you know?
Found some dual SID $D420 music and that seem to work.
Hope to hear from you soon. a big thx already !

vanfanel, 2024-06-12:
@vossstef With this new .bin, I could finally test the new SID options.

The only filter setting that sounds correct to me is "Follin".
Extreme sounds too raw. I believe "Follin" is the default, right?

The reference game is Ghouls'n Ghosts, not Goblins, because Ghoulsn'n Ghosts music is by Tim Follin and does very sophisticated use of filters.

vossstef, 2024-06-12:
@vanfanel the default filter in reDIP-SID is 'average'. 
Please store those setting in OSD.
 
Do you think that the new Design is suitable and good enough in terms of SID sound ? I Don't need now an answer. If you tell me let's say in 2 days is also fine. Still need to fix something related to serial RS232 Interface. thx

vanfanel, 2024-06-13:
@vossstef I have been testing both SID implementations all morning now (by going back and forth between stable .fs and this branch's test .fs) and I like the sound in the MiSTer version better, specially for Tim Follin games.
Thankfully, things are quiet on my workplace, so I have time for these delicate tests.
I brought my hi-quality headphones with me, to do the tests as they deserve.

I will be doing more tests with demos and other games, and will try to provide a definitive answer in some days.
I think this question needs to be carefully considered.

I have also opened a thread on the MiSTer forum to request opinions:
https://misterfpga.org/viewtopic.php?p=87479#p87479

...But for now, I would say the MiSTer version sounds better to me, more akin to real C64, also in Ghosts'n Goblins.

vossstef, 2024-06-13:
Thanks for your effort !

Did you use the 'average' setting for your comparisson test? 
I can't tell at the moment which coefficients are implemented.  maybe i nedd to dig a bit in MISTer. Will let you know if i Figuren out

vanfanel, 2024-06-13:
@vossstef Yes, but the "average" sounds pretty off compared to the MiSTer implementation.

I have also asked Sorgelig, the MiSTer creator, and he answered that the MiSTer implementation is more accurate, which confirms what I hear when doing these comparisions...

vossstef, 2024-06-13:
@vanfanel looks that i need to rework the code significantly respectively need to make use of the latest redip-sid varaint. I only got a version running that is about similar age that Mister sid. Lokks that Mister added some more tweaks that i am missing now. Anyway i learned a lot and will release an intermediate step just to fix DB9 Joystick and Paddle for now.

vanfanel, 2024-06-16:
> @vanfanel i got the dual SID based on well knows MIST'er core finally done and just created the release. Pls. let me know if there are problems.

Ok, I will test as soon as I can! Thanks!
I am currently traveling, so I will start testing on wensday as soon as I am back at home :)

vanfanel, 2024-06-18:
> @vanfanel i got the dual SID based on well knows MIST'er core finally done and just created the release. Pls. let me know if there are problems.

Ok! I've been testing the latest stable version, and 1-SID chip (both 6581 and 8580 models) sound perfect to me with games and demos.

Howevel, dual-8580 seems to be missing one SID, or it sounds bad. For dual-8580 reference, I use this astonishing demo:
Here you have it in .D64 format, since current version of the core doesn't seem to load .PRGs properly:
[LMan - Blade Runner Main Titles.zip](https://github.com/user-attachments/files/15890559/LMan.-.Blade.Runner.Main.Titles.zip)

Here is how it sounds when played properly (sounds perfect on the MiSTer C64 core):
https://www.youtube.com/watch?v=XpP9z9gEMXA

vossstef, 2024-06-18:
@vanfanel , thansk for all your effort and will check on that. 
I got in the meantime the SID Filter coefficients loader to work and will add it soon. For my bad ear's it is close to impossible to hear big differences... Unfortunately the FPGA is now entirely full and i had to sacrifice the PAUSE ON OSD feature.

vanfanel, 2024-06-18:
@vossstef Are PRGs supossed to be working? Some demos are in that format, but as I said loading PRGs seems to be broken (I don't want to open another non-ussue on that, but couldn't find any information on the docs about loadinr PRGs)

vossstef, 2024-06-18:
@vanfanel , pls. let me check on that with the stored bitsteam in the release section. I will let you know tomorrow.

vanfanel, 2024-06-19:
@vossstef Please disregard my previous message saying that dual-8580 does not sound correct. It DOES sound correct.
The Blade Runner demo seems to need dual-8580 from the begining and digifix ON, then it sounds as expected.
So I consider dual SID audio perfect now on the core.

Other dual-SID demos you can enjoy:
https://csdb.dk/release/?id=219599
https://csdb.dk/release/?id=157533

vossstef, 2024-06-19:
@vanfanel thanks for feedback. Don't worry as i didn't found a second to look into that.
Will let you know when the SID filter load feature is ready for testing.

As it stand now will have today no time for PRG load testing. Tomorrow...

vossstef, 2024-06-20:
@vanfanel i right away release a new version with the SID FIlter curve load feature. 
PRG load is working on my side.
Let me know if you find buys. thx

vanfanel, 2024-06-21:
@vossstef This new version does not work correctly here. Loading CRTs or D64s, the core seems to hang. Nothing seems to work correctly. Maybe it was a bad build somehow??

vossstef, 2024-06-21:
Hi @vanfanel ,
could you pls try to delete the c64.ini on the sdcard and after boot. Set all execpt the d64 to 'no disk' and save settings.  you habe to program the New firmware too. i flashed my board yesterday from the stored bitsteam and it worden correctly. Donut fully unverstanden what the problem is.

vanfanel, 2024-06-21:
@vossstef Yes, I tried deleting c64.ini, power-cycle, set all except D64 to "no disk", and save settings.
I also updated the firmware.

But really, nothing loads. C64 boots correctly, but then loading D64s, CRTs, etc.. all fails.

Even doing a simple `LOAD "$",8` with an inserted .D64 fails (the machine never completes the task).

vanfanel, 2024-06-21:
@vossstef Going back to version 1.2.3, everythig works as before.
So version 1.3.0 is clearly bugged.
I do these tests on the Tang Nano 20k (not 25k or 138k)

vossstef, 2024-06-21:
I just programmed again my TN20k and bl616 and it works fine.
Tested also TP25k and working fine.

Need to find another TN20k and try on that.

vossstef, 2024-06-21:
hi @vanfanel ,
i checked another another TN20K , Dock M0S combination and that is working fine too. a 3rd tn20k will test later on.

ojaksch, 2024-06-21:
Can copy that in _some_ way, but I'm not familiar with a C64. CRTs are working fine with JiffyDOS, **not** with DolphinDOS. D64 aren't working in any way.
But there's something weird: No matter what c1541 ROM I'm using, it's booting with DolphinDOS (except JiffyDOS, which is working fine - w/o D64's)
Have to reflash all DOS'es tomorrow and doing a re-test then.

tn20k and bl616, both flashed with the new 1.3.0 files today.

vossstef, 2024-06-21:
@ojaksch will check on that later on.

ojaksch, 2024-06-21:
Couldn't wait 'til tomorrow :) My ROMs were all 32k (old documentation?) and I've re-flashed them with the now correct size, according to the current instructions. Now all four slots are booting with DolphinDOS and nothing loads (CRT, D64) anymore.
Is there a way to "reset" (clean) all flash areas on the tn20k? Is it okay to flash fs and ROMs while M0S is connected to the PCB (but flashing via USB-C/tn20k)?

vossstef, 2024-06-21:
Hi all,
i got the core on three different TN20k running.  
It's all about the content of the sdcard. 
delete the c64.ini 
and have following files installed: 
disk8.d64
c64flt.flt
c64kernal.bin
c64prg.prg
c64crt.crt
c64tap.tap 

I didn't check the different DOS rom swicthing yet but default Dolphin Kernal + c1541 DOS is working fine.

vossstef, 2024-06-21:
Hi all,

guess i found the issue. I missed to introduce a change in the misterynanofw... (mountpoint for new .flt).
will fix now and upload a new fw. soory !

vossstef, 2024-06-21:
@vanfanel and @ojaksch @harbaum i just released an updated firmware fixing the missing mountpoint for FLT drive.

vanfanel, 2024-06-21:
> @vanfanel and @ojaksch @harbaum i just released an updated firmware fixing the missing mountpoint for FLT drive.

I have updated the firmware, deleted the c64.ini file, copied over c64flt.flt to the root of the SD... but still I see CRTs and D64s are failing to load correctly. Nothing really works.

vossstef, 2024-06-21:
@vanfanel i also found a malfunction in case no disk8.d64 is selected. Propose to have at least the image for c1541 always selected. CRT and D64 is working fine on my side.

vanfanel, 2024-06-21:
> @vanfanel i also found a malfunction in case no disk8.d64 is selected. Propose to have at least the image for c1541 always selected. CRT and D64 is working fine on my side.

What's disk8.d64? Is it an special D64 or will any D64 do?
I don't understand... if I have disk8.d64 selected, how will I load my own D64 files?

vanfanel, 2024-06-21:
Ah, I see. Any disk will do as disk8.d64.
However, after putting disk8.d64 on the root of the SD, and selecting it in STORAGE, D64s and CRTs fail to work here.

vossstef, 2024-06-22:
@vanfanel and @ojaksch i think i found another issue that was introduced by the FLT loader (6 drive support).
I just uploaded a new bitstream for TN20k. hope that it helps.

vanfanel, 2024-06-22:
> @vanfanel and @ojaksch i think i found another issue that was introduced by the FLT loader (6 drive support). I just uploaded a new bitstream for TN20k. hope that it helps.

I'm sorry to say that loading D64s and CRTs also fails with this new TN20k bitstream.
Please, consider going back to 1.2.3 version, this is broken.

ojaksch, 2024-06-22:
@vanfanel While Stefan is investigating into this, try to use jiffydos (or dolphindos) as Kernal and jiffydos as 1541 ROM. This is the only combination which is working for me with D64s and CRTs.

vanfanel, 2024-06-22:
> @vanfanel While Stefan is investigating into this, try to use jiffydos (or dolphindos) as Kernal and jiffydos as 1541 ROM. This is the only combination which is working for me with D64s and CRTs.

Ok, do you mean using the same Jiffydos ROM file as kernal and 1541 ROM?

vanfanel, 2024-06-22:
@ojaksch Ah! Got it to work by simply selecting JiffyDOS in c1541 ROM.
With that, D64s and CRTs seem to load fine.

ojaksch, 2024-06-22:
@vanfanel Looks like that something at our site is different to Stefan's. We'll see...

vossstef, 2024-06-22:
@vanfanel and @ojaksch i scrapped the release for the time being and will re-release when i find time. sorry for the trouble caused. Need more time for proper analysis.

vanfanel, 2024-06-22:
> @vanfanel and @ojaksch i scrapped the release for the time being and will re-release when i find time. sorry for the trouble caused. Need more time for proper analysis.

Don't worry at all!
I will be happily testing future versions, this core is awesome and I can't thank you enough for all the work you put on this.

ojaksch, 2024-06-23:
> Don't worry at all! I will be happily testing future versions, this core is awesome and I can't thank you enough for all the work you put on this.

I copy that! No worry, no hurry! We are here to test it for any glitch or quirk we can find and are happy to help.

vossstef, 2024-06-25:
@vanfanel , @ojaksch  think that i found the bug related to the malfunction of the loader... 
It was related to bl616 firmware. I missed to change one bitfield when i updated sdc.c to support for 6 drives. In addition for unknown reasons (maybe IDE version change to .03 ???) a problem with the sdcard flash ROM module.
I corrected firmware and the tn20k bitstream. Stored both in the main  branch for the time being (no new official release before more testing...)  Sorry but i didn't found time to test the SID filter load function properly yet. At least it looks that it does not disturb other functions like d64 / prg /crt this time.

vossstef, 2024-06-25:
I check the effects of the SID filters with a program called PSID64 1.0 Roland Hermans. 
For the very first time i was able to check the differences in between all the different filters settings and also the loadable filter (It works ! ) that i stored in the branch (custom.flt and Alankila6581R3_3984_1.flt)

ojaksch, 2024-06-26:
Um, I'm begging for pardon but can't see any difference with the new "release". Everything is still working (fine) as long as I use jiffydos in the 1541. Am I missing something?
Did only some quick tests, a demo (d64) here, a game (d64 and crt) there...

vossstef, 2024-06-26:
the 1.3.0 release was buggy for sure. In the bitstream and firmware i  stored temporary directly in the main brach there should be at least the Bugs related to the new FLT filter load be gone( the 6th drive). About TAP load i feel unsure as not tested. i tested default dolphin c64 kernal and c1541 dolphin DOS and that was fine loadimg d64 images. jffy i didnt test so far.

vossstef, 2024-07-09:
@vanfanel and @ojaksch i did a 1.3.1 release.
Most important is that the TAP autoload is removed and README enhanced with lot of hopefully helpful information.
I checked D64, PRG, TAP, FLT, BIN and combinations and hopefully it is working better now. Buildin Dolphin Kernal is working fine as usual. Speeddos and jiffy didn't test yet.

May i propose that you start testing but initially set all loaders to 'no Disk' and enable one after another or better more and more.

vanfanel, 2024-07-10:
@vossstef I tried this 1.3.1 release today, and I did what you said:
CRT loading is broken. Try MultiLoad games, they don't work (simple SingleLoad CRTs work, but that's all that works).

vossstef, 2024-07-10:
@vanfanel sorry!
i forgot to test multiload crt....

TAP after CRT load is not working for exbasic II but that is the only Bug i found.
will habe this evening a look

vossstef, 2024-07-10:
@vanfanel ,
i just updated in the 1.3.1 release the tang_nano_20k_c64.fs file.
CRT multiload is working now.

vanfanel, 2024-07-10:
@vossstef Ah, yes, yes, thanks! :)
CRT (single and multiload) works fine.
D64 works fine.
TAP works fine.
PRG does not work correctly (it never did, so it has not broken now, try to load this for reference: https://csdb.dk/release/?id=138582)

May I suggest a couple of things?
-Can you add a key to eject the inserted media? For example, MiSTer uses BACKSPACE.
-Please move joystick-on-keyboard keys to ARROW KEYS + LEFT CTRL?

vossstef, 2024-07-10:
thanks
will check agin prg

vossstef, 2024-07-10:
Hi @vanfanel ,
i am using exactly the  bladerunner PRG you listed above to check the dual SID function. 
It is working like a charm over here. 
I use in addition Singularity https://csdb.dk/release/?id=135891&show=notes
Also  working like a charm over here. 

Can you pls check with after PRG load with LIST +CR what is there ? Then do RUN + CR

BUT i realized in case if you set for the c1541 'no disk' and do OSD cold boot or Power cycle the c1541 sdcard loader get stuck and will not load any PRG. 
propose till i figure out how to fix alway leave the D64 properly set as boot8.d64 or other D64. It will anyway not harm.
Can you pls. let me know ? Thx !

vanfanel, 2024-07-10:
@vossstef I confirm that I can use LIST and RUN with a PRG, but only if a .D64 is inserted. So yes, I am seeing the same bug :)

[7 more replies]