MiSTle-Dev/C64Nano #57: MultiLoad games from the C64 OneLoad collection (.CRT files) fail to run.

vanfanel, 2024-05-17:
That's about it: the MultiLoad games in the C64 OneLoad collection (they are .CRT files) fail to load.

The C64 OneLoad collection v5 is the most famous C64 games collection and can be downloaded here:

https://mega.nz/file/WZUV2ApA#JFoFB3vVCcfiKe02TkTqq3JOR2OoWLWBGA2B8m_bxbI

vossstef, 2024-05-17:
let me check on that that next week. So far i tried single files only.

vossstef, 2024-05-22:
Bug confirmed.
solution available. To be included in next release

vanfanel, 2024-05-22:
Great! I'm very glad to read that! In fact, it will make massive game testing way faster! :)

vossstef, 2024-05-24:
@vanfanel 
is also this topic solved in the branch i created for testing?

vanfanel, 2024-05-24:
@vossstef Yes, it's solved on that branch! 
In fact, I used that to test many of the games in CRT format! :+1: 

So I believe this can be closed, right?

closed 2024-05-24 by vossstef