MiSTle-Dev/C64Nano #60: Image not centered in PAL mode

vanfanel, 2024-05-25:
Hi there again, @vossstef 

This is a relatively small problem, but it's there and it would be nice to have image properly displayed.

On all my HDMI monitors, the core image is not centered: it's too low, as seen in this picture:

[image]

So, maybe it has something to do with timings? Could it be centered?

Thanks!

closed 2024-05-25 by vossstef

vossstef, 2024-05-25:
i fixed that an hour aho. pls check

vanfanel, 2024-05-25:
@vossstef Tested! Image is perfectly centered in PAL mode now!!

[image]

However, I noticed NTSC image is not centered (this time horizontally):

[image]

Is the NTSC image also fixable?

vossstef, 2024-05-25:
let me habe a Look into that

vossstef, 2024-05-28:
@vanfanel 
i tried endless different hdmi settings in the NTSC mode and after all it looks that better leave "as is". sorry...

vanfanel, 2024-05-28:
> @vanfanel i tried endless different hdmi settings in the NTSC mode and after all it looks that better leave "as is". sorry...

Don't worry. I Will use the core in PAL mode anyway, I am european and don't care much about NTSC software. 
Many thanks for trying!! I am sure you put a lot of time in trying values... And time does not come back.