MiSTle-Dev/C64Nano #70: USB Joystick behaving strangely

BennehBoy, 2024-06-18:
Hi,

I figure this is probably an upstream issue with the BL616 code, or potentially a strange USB hub interaction. If I connect up an 8 pin D joystick via a D to USB adapter then game input when the joystick is at rest has down and right continually selected.  Any actual input overrides this.  I also have a HUB in this equation so it may also be introducing an issue but the combination works fine with other projects.

This joystick/adapter combination works fine on linux/windows/rpi but I'm unable to test it with the HUB circuit with the tang.

The adapter I'm using is homebrew -> https://github.com/BennehBoy/ZipstickUSB 

VID/PID : 040b:6533

I've not encountered issues with it previously.

A generic USB SNES gamepad works without issue.

VID/PID : 0810:e501

harbaum, 2024-06-18:
I'd assume this is some misinterpretation of signed vs. unsigned axis reports sent by this joystick. I thought I had sorted this a long time ago ... may need another look ...

BennehBoy, 2024-06-19:
I noticed that the firmware for the BL616 included in the nanomig core has a larger file size and a newer datestamp so also tried that - although everything seems to function fine, the joystick issue is still there.  Is the nanomig 616 firmware specific to that core?