MiSTle-Dev/C64Nano #80: MISS keys triggering delete

SanguineBrah, 2024-07-25:
I'm not sure if this is an issue with FPGA companion or the core, so apologies if I am posting in the wrong place.

In my testing, any key marked as MISS in the C64 keymap table seems to be interpreted as MATRIX(0,0) by the core, which is the delete key.

However, this is not the case with the modifier keys, which just (correctly) do nothing if they are marked MISS.

vossstef, 2024-07-27:
Hi @SanguineBrah ,
i updated the firmware and swapped in the release section of c64 and vic20 core.

Pls have a try and report. 
Thx for feedback to improve the core !

SanguineBrah, 2024-07-27:
Thanks. I've tested the new firmware on the release page and a lot of keys now trigger a left shift event, including HOME, END, PGUP, PGDN, BREAK, NUMLOCK, SCRLOCK, TAB and PRINTSCR. The restore key (F11) also triggers a left shift at the same time as successfully triggering a restore event.

For testing, I am using [Anykey](https://github.com/T-Pau/Anykey).

vossstef, 2024-07-27:
i implemented left shift as workaround!

is that fulfilling your needs? 

left shift didn' produced  unwated  Charakters on the screen

SanguineBrah, 2024-07-27:
I don't think shift+restore does anything so it should be okay as a workaround, thanks.

vossstef, 2024-07-27:
@SanguineBrah ,
i agree.
Restore is bare NMI trigger if i recall right. Didn't' tested all those 'freezer' CRT yet....

closed 2024-07-27 by vossstef

harbaum, 2024-07-29:
Indeed, the MISS code is sent to the core as 0 and if 0 actually is a valid code then this triggers something. 

This is something I'll address in the new FPGA Companion firmware asap ....