MiSTle-Dev/C64Nano #81: CRT not loaded at first powerup

SanguineBrah, 2024-07-28:
On first boot, c64crt.crt shows as loaded in the Storage menu but it isn't loaded in the core. The Reset option doesn't fix it but using the Cold Boot option then successfully loads the default cartridge.

I am testing using the CBM kernal and the [Epyx fastload cartridge](https://rr.pokefinder.org/wiki/Epyx_FastLoad) (FASTLOAD should show after the READY prompt).

vossstef, 2024-10-18:
Multiload CRT got once again broken...

vossstef, 2024-10-18:
I just created a new release that corrects the problem. 
Pls try and report. thx

SanguineBrah, 2024-10-18:
I'm seeing the same behaviour in the latest release. On first boot, the default cartridge is not loaded. It still isn't loaded on using the reset option but using the cold boot option in the menu then loads the cartridge correctly. The cartridge is also loaded correctly if I select it from the menu.

vossstef, 2024-10-18:
@SanguineBrah, confirmed! 
Looks like that also normal CRT load via SD card after power-up got broken. 
It works if you load normally after power op via OSD.

vossstef, 2024-10-18:
@SanguineBrah, after extensive testing i figured out that the power-up CRT load issue is not present with the misterynano fw.
I updated misterynano with latest c64 and vic20 core OSD needs and added the .bin and cfg to the release also.
Would be great if you will find time to test that and please let me know.

SanguineBrah, 2024-10-19:
@vossstef I can confirm that the misterynano firmware in the new release fixes this issue for me. Cartridges are now loading immediately on first boot and behaving correctly.

harbaum, 2024-10-19:
Please feel free to do Companion releases yourself. If this temporarily  breaks another core then this will just be fixed asap.

vossstef, 2024-10-19:
great!
Now need i to find out and fix for companion.

vossstef, 2024-10-19:
@SanguineBrah,
i created a 1.6.0 pre-release supporting apart from proper Multcard load also a freeze function with e.g. Action Replay CRT.
Not sure is Epyx fastload also supporting freeze but feel free...

vossstef, 2024-10-20:
CRT issues fixed in release v1.6.1

closed 2024-10-20 by vossstef