MiSTle-Dev/C64Nano #83: Feature suggestion: serial over spare I/O

SanguineBrah, 2024-08-13:
The RS-232 over USB works well, but I have found that its usefulness is limited because it requires a host device on the other end, which means it is only good for a null modem link with a PC.

If it is possible to put this out on spare I/O pins instead, we could connect it through a MAX3232 for a standard serial port, wire up an FTDI adapter module for USB connection or wire a header for an [ESP-01 wifi module](https://www.microchip.ua/wireless/esp01.pdf) for onboard wireless telnet using the [zimodem firmware](https://github.com/bozimmerman/Zimodem).

harbaum, 2024-08-13:
For telnet there's actually a way nicer solution. The latest firmware runs on the pi pico-w and the esp32 as well and these have wifi built-in.

This is not released yet, but I am using these with the Atari st core to redirect rs232 via wifi so I can dial via Internet into BBS' that way. This gives quite a retro networking experience.

vossstef, 2024-09-10:
@SanguineBrah i added an option for hw uart pin interface . 
Do you want to test it ?

SanguineBrah, 2024-09-10:
> @SanguineBrah i added an option for hw uart pin interface . Do you want to test it ?

That's great! I'll take a look at it tomorrow.

SanguineBrah, 2024-09-11:
I have done some testing using CCGMS with the RX & TX wired up to a Wemos D1 Mini with wifi modem firmware. Everything works as expected. I am able to reach up to 2400 baud without dropping any characters in either UP9600 or Vicmodem modes - to reliably go faster, we would probably need flow control, which would use up two more spare I/O pins (it is arguable whether that would be worth it). I was not able to get the Swiftlink modes to send any data but I am not familiar with that device so I may be missing something there.

vossstef, 2024-09-11:
> Yes, I tested on 20k as I don't own any of the others. I haven't seen any side effects or stability issues so I think it should be okay to merge if the other models are also working the same way as this.

ok then will merge and create a draft release for easier use. Still need to consult @harbaum  to agree pinmap for TN25k/TM138k PMOD. share.

harbaum, 2024-09-11:
The pin sharing sounds fine to me. I should have broken out the unused pin on the pmod adapter ...

closed 2024-09-11 by vossstef

vossstef, 2024-09-11:
@SanguineBrah i created a pre-release and dropped the bitstreams
readme is also updated