MiSTle-Dev/C64Nano #89: Issue with games that use paddle inputs as extra buttons

SanguineBrah, 2024-09-29:
I tried out [Gates of the Ancient](https://drmortalwombat.itch.io/gates-of-the-ancient) today and found that it is currently unplayable with this core when using a DE9 joystick (in this case, the ArcadeR plugged into harbaum's midi shield).

This is because it uses pins 5 and 9, which are paddle inputs, as extra buttons. On the Nano, the extra button reports as permanently depressed, which activates the cockpit menu and means you can't steer the ship.

I also found out that there are [a few other commercial games](https://retrocomputing.stackexchange.com/questions/2499/which-c64-games-support-two-buttons) which used the paddle inputs in this way.

vossstef, 2024-09-29:
will look into that over the next weeks

vossstef, 2024-10-04:
Hi @SanguineBrah,
could you please test your game with my latest release ?
I checked with a diagnostic tool the 2nd and 3rd button but are not sure is the function properly for a game implemented.
thanks!

SanguineBrah, 2024-10-04:
In Gates of the Ancient, the behaviour is the same. I tested further using [Game Controller Tester](https://commodore.software/downloads/download/148-c64-testers/11748-game-controller-tester-v1-0), which shows that all POT values are at zero and the 2nd (64GS) button is permanently activated on both ports. Pressing the 2nd button on my joystick doesn't change the reading. By opening my joystick and putting it into Atari mode, it grounds the pin instead of putting 5v on it - in this case, when I press the button, the tester shows the 2nd button being released and the POT reading jumps to 255.

I tested on my real C64 to compare the behaviour (joystick back in C64 mode). Using the Tester program on the real machine, all POT values show as 255 and the 2nd button is not activated on either port. When I press the 2nd fire on my ArcadeR joystick, one POT drops to 2 and the 2nd fire button shows as activated in the tester. I also tested this joystick on Gates of the Ancient and the 2nd fire button is working correctly there on the real machine.

vossstef, 2024-10-04:
@SanguineBrah thanks for Feedback!

will also use this test sw to figure out and check in moment detail.

in which mode do you use the ArcadeR? i saw that there are pcba solder bridge fitting options for different computers.

the misteryshield and my core is only compatible to atari st 2 Button Joystick types meaning that a closing  switch in between d9 pin 9 and GND is expected. C64 Mode (pin9 and +5v) or atari 2600 vcs will not work.

SanguineBrah, 2024-10-04:
I've tested it in both. It looks like the Atari mode (connecting to GND) responds but the behaviour is inverted - i.e. POTs all default to 0 (2nd button active) then show as 255 (2nd inactive) when pressed.

I have also tested using a classic Atari 2600 joystick with only one button and the 2nd button shows in the tested as being permanently active (POTs at 0).

vossstef, 2024-10-04:
@SanguineBrah 
i found the bug and it was inversion of the analog data. 
Fixed now in a new release **v1.5.1**
As mentioned for Tang D9 port use the Atari ST mode http://wiki.icomp.de/wiki/DE-9_Joystick of ArcadeR

SanguineBrah, 2024-10-04:
@vossstef Thanks, v1.5.1 seems to be working great! Everything looks good in the tester with my joystick and Gates of the Ancient is playing correctly just like it plays on my C64.

closed 2024-10-04 by vossstef