MiSTle-Dev/C64Nano #102: everything is a bit slower/lagging in V1.6.2

sgrockfort, 2024-10-26:
Hello, just noticed a bug in 1.6.2 .... everything is a bit slower/lagging... Sound and movement are lagging.... This happens in PAL & NTSC mode... I tried it with "Boulder Dash 1".. Problem did not appear in v.1.6.1

Thanks
Michael

vossstef, 2024-10-28:
Hi @sgrockfort, i created a 1.6.3 pre-release to fix the lag issue...
pls. have a try...

sgrockfort, 2024-10-29:
@vossstef  , perfect thank you... I can try it today evening...

sgrockfort, 2024-10-29:
hi @vossstef  , just tested the 1.6.3 ... Switching between pal & ntsc works fine.... gaming in PAL mode also works good.. I am not 100% sure!!!  but in PAL mode, it feels a very tiny bit faster as usual - also it seems that the game speed in PAL and NTSC is equal... Usual NTSC always a bit faster.....  But it's ok imho.

Also I noticed that in NTSC mode the sound seems to be broken.... The sound is interupted every 1-2 seconds by a very soft click sound... This sound is even noticable directly in the C64 start screen... In PAL mode the sound is fine...

I made my tests with standard settings - no filter files or something like that.
For game-test I used Boulder-Dash 1 once more :-)

vossstef, 2024-10-29:
@sgrockfort Thanks!
I tested a lot in PAL mode but obviously too little in ntsc...  Those sound issue i didn't detected. Will have again a look and let you know.

vossstef, 2024-10-29:
Hi @sgrockfort , 
i was also able to detect the noise but not the speed topic.
i synthesized again with IDE 1.9.10.03 and a buildscript. Checked with one Game and the clicking noise in NTSC seeme to be gone. 
Stored the file as pre-release v1.6.4
Maybe could you also check... big thx !

sgrockfort, 2024-10-30:
Hi @vossstef  , no problem.. I'll check it later.. Thanks for fixing...

sgrockfort, 2024-10-30:
@vossstef , good evening, I flashed the v.1.6.4, but in NTSC mode I still have the sound issue.. I attached a short audio record how it sounds..  I flashed it twice, just to ensure that I didn't made a mistake.. 

[Neue Aufnahme.zip](https://github.com/user-attachments/files/17577592/Neue.Aufnahme.zip)

vossstef, 2024-10-30:
@sgrockfort thx for feedback. A  nightmare...
FPGA is entirely full and i assume a malfunction due to that. Will try to exclude certain function e.g. MIDI to free some resources and try clock constrains... Will try to fix.

sgrockfort, 2024-10-31:
@vossstef  , just let me know if I can help with testing.. Completley no problem, I just need some time sometimes ;-)

vossstef, 2024-11-01:
@sgrockfort 
created now a v1.65 pre-release....

From early tests it looks better but let's see..

sgrockfort, 2024-11-01:
p.s. also tried with other sound chip settings.

vossstef, 2024-11-01:
@sgrockfort thanks !
i tested a single tn20k board and also a tp25k using different games (also Boulder Dash) and sound was always fine both in PAL and NTSC mode. Tested also G64 disk, RS232 with terminal , MultiCRT and freezer all was ok.

Will flash an another tn20k to figure out those sound noise topic...

sgrockfort, 2024-11-01:
@vossstef  , hmm, maybe it's something specific to my setup... I just also noticed that this click sound also appears when the volume is completley muted.. Also noticed that the gamesound gets very quiet ini the rhythm of the click.... I think what I hear is the result of switch off the sound off/on in this rhythm... It only appears in NTSC mode - that's why I thought its something about my screen. But with 1.6.1 it's okay.

But if you cannot reproduce it, it's maybe really something related to my setup and as for me no need to change anything.. In general I only use pal :)

sgrockfort, 2024-11-02:
@vossstef  , I can confirm that this problem is related to my screen.. I switch my screen to my old 1080p screen and there is no sound issue in NTSC anymore... I still cannot understand why it not happen with v.1.6.1 but anyway... Maybe it's something with upscaling ?! (new Screen is 4k)

Thank you for helping and sorry for the confusion.

sgrockfort, 2024-11-02:
@vossstef  ,  just an addional info - I found out that with v1.6.1 the screen-resolution in NTSC is 800x480@60Hz and with 1.6.5 it is 800x480@58Hz (if my screen shows the values correctly) .... Maybe my screen is just a bit sensitive about it and not tolerant about it...

vossstef, 2024-11-02:
@sgrockfort i found a solution now and still testing the correction. Looks that the DCS clock management is causing the trouble. new fix is simple not needing that.😀

vossstef, 2024-11-02:
@sgrockfort , i created a v.1.6.6. based on the working 1.6.1

sgrockfort, 2024-11-02:
Hello @vossstef  , I'm sorry but in my environment the NTSC-sound is still broken. It seems to be a very specific behavoir related not to all screens.. As for me, no need to change it. But if you want/like to continue work on it, I can continue testing. Completley no problem... :-)

vossstef, 2024-11-02:
I checked many games and including Boulderdash and SID sound reference Ghost'n & Goblins in NTSC and PAL mode using HDMI to USB converter and not too old Dell 4K monitor without any noise problems. G64 Disks are now working, RS232 is fine and MultiCRT ROMs, freezer CRT, Kernal, CRT and PRG load after power-up is working. 
Maybe your noice issue is specific to your setup.

I found malfunction if the OSD XML file is missing on the sdcard.

sgrockfort, 2024-11-03:
@vossstef  , yes, I also think so.. It is something about my specific screen.. As I said, with my old/other screen I haven't any problems also.. I think this screen don't come clear with the 58Hz..

Thanks once more for all you work. I really like your project.. If you need a tester for other future problems - no problem :-)

Michael

vossstef, 2024-11-03:
@sgrockfort , 
all fine and thanks for all the effort in testing and news that your other screen works fine !

closed 2024-12-01 by vossstef