MiSTle-Dev/C64Nano #116: Adding support for more USB keyboards, possible?

vanfanel, 2024-12-03:
Hi there,

I've been happily using this absolutely fantastic C64 core for some time now, and recently tried a mechanical 8bitdo keyboard with it. It didn't work. Same with another mechanical QNIX keyboard I have.

However, my old trusty (but not so pleasant to use) Logitech keyboard+mouse combo works perfectly well with it!

This is how GNU/Linux's `lsusb` sees the working keyboard:
`Bus 001 Device 005: ID 046d:c534 Logitech, Inc. Nano Receiver` 

And this is how it sees the non-working 8bitdo mechanical keyboard:
`Bus 001 Device 006: ID 2dc8:5201 8BitDo 8BitDo Retro Keyboard Receiver`

So, I guess this depends on the FPGA Companion (bl616) firmware, right?
Where in that firmware should I add new keyboard identifiers? Can that firmware be rebuilt on GNU/Linux easily? I hope so! :)

vossstef, 2024-12-03:
Hi @vanfanel ,
great to hear from you again and i am very happy that you are still using the core !
Let's ask @harbaum what could be a blocking factor for this specific Keyboard to see is there a fix possible in the FW.

camplusplus, 2025-03-08:
I also have USB keyboard (with joystick) issue, reported to https://github.com/harbaum/FPGA-Companion/issues/15

camplusplus, 2025-08-25:
I found solution by converting usb to ps2 with esp32 and using ps2 to usb convertor.

[image]