MiSTle-Dev/C64Nano #124: Can't update the C64 core on Tang Nano 20K flash anymore. Can you lend a hand, please?

vanfanel, 2025-02-10:
Hello there @vossstef 

I have been updating my TangNano 20K to the latest C64 core via the USB-C cable for a year or more.
I was trying to update to the 1.8.0 release today, and I am getting errors I had never seen before:

```
manuel@debian:~/C64$ sudo openFPGALoader -f tang_nano_20k_c64.fs
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse tang_nano_20k_c64.fs: 
Done
DONE
after program flash: displayReadReg 00004421
        CRC Error
        Memory Erase
        Non-JTAG configuration is active
        Security Final
Erase SRAM DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
RDSR : ff
WIP  : 1
WEL  : 1
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Error: write to flash failed
Error: Failed to program FPGA: Error: write to flash failed
```

And then, if I pass the "--unprotect-flash" parameter, I get:

```
manuel@debian:~/C64$ sudo openFPGALoader -f tang_nano_20k_c64.fs --unprotect-flash
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse tang_nano_20k_c64.fs: 
Done
DONE
after program flash: displayReadReg 00004421
        CRC Error
        Memory Erase
        Non-JTAG configuration is active
        Security Final
Erase SRAM DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
RDSR : ff
WIP  : 1
WEL  : 1
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
timeout: ff ff ff
ff
wait: Error
Error: write to flash failed
Error: Failed to program FPGA: Error: write to flash failed
```

This is on GNU/Linux, of course.
Do you have an idea on what could be going on, please?

vossstef, 2025-02-10:
hi @vanfanel, looks to me that something wrong with the openfpga loader...
i also had problem with that tool and i use now the 1.9.10.03 gowin Programmer sw instead also under Fedora.
had also the feeling after update to f40 / f41 problems got more accute. 

you need to run the gowin programmer as sudo otherwise it will fail at some point.

gowin programmer also has issues with "stoneold" GW1 devices... i had to use for TN9 very old toolset to get that programmed.

vanfanel, 2025-02-10:
Thanks @vossstef 

I'm trying to use the GoWin programmer but it's VERY confusing. Where is the option to flash the .fs files to the tangnano20k board? I can't seem to find a way to open .fs files or flash them...

vossstef, 2025-02-10:
@vanfanel pls have a look into Till's or my instruction document how to use the programner

vanfanel, 2025-02-11:
> [@vanfanel](https://github.com/vanfanel) pls have a look into Till's or my instruction document how to use the programner

Thanks, looked in the docs and found the precise instructions.
I hope OpenFPGALoader works again someday.. it was way better and easier on the Linux CLI.
The graphical Gowin programmer is clunky to say the least...

Many thanks! Now enjoying the latest core version again! :)

closed 2025-02-11 by vanfanel

vossstef, 2025-02-13:
@vanfanel 
@harbaum 

just saw this !

https://dl.sipeed.com/shareURL/TANG/Debugger/onboard/BL616/202501231x

looks they found issues and fixed something...

harbaum, 2025-02-13:
Please be careful when using these. I had trouble with different tn20k versions when some of them needed  signed/encrypted firmware versions and some did not. 

My recommendation is to only update when you really have a problem and not just because "newer is better".