MiSTle-Dev/C64Nano #135: TheC64 + tang_nano_20k_c64: How to send the menu open command?

vanfanel, 2025-04-07:
Hi there!

I don't know it someone had tried before, but I connected the TheC64 keyboard to the TangNano20k+misteryshield20k, and it woks perfectly well.
(TheC64 is a full size C64 replica running an small GNU/Linux system with the VICE emulator on top: https://www.c64-wiki.com/wiki/THEC64) 

[image]

...So my dream of gutting a TangNano20k+misteryshield20k inside TheC64 is near.
However, since the C64 has no F12 key, there's no way to enter the core menu.

Is there a couple of pins somewhere on the TangNano20k or on the misteryshield20k that I can use to attack a switch to them so I can enter the core menu with that?

Thanks!

vossstef, 2025-04-08:
Hi @vanfanel ,
the F12 key is something special in the Fpga companion fw. you will need a customization in that. never tried before but you can transfer the two tn20k button state into the fw. hope my c64 is doing that already and ifnot will add.that. bigger part is to modify fw that one of the two received buttons is equivalent to F12. All maybe possible. btw one of the buttons is already in use to swap joyports and other one if i recall unused. sorry for the short answer but rather busy at the moment.

vanfanel, 2025-04-08:
> Hi [@vanfanel](https://github.com/vanfanel) , the F12 key is something special in the Fpga companion fw. you will need a customization in that. never tried before but you can transfer the two tn20k [...]

Many thanks for the response! Not in a hurry at all, don't worry :)
Isn't it possible to use a couple of pins on the companion board? It has 8 pins (which I don't know what they do, to be honest, and I can't find any information on what they do).

vossstef, 2025-04-08:
@vanfanel , you could also use one of the spare FPGA pins that i don't use for the DS2 interface for a Key F12 button. But the issue of forwarding the pin press into the firmware will remain. There is some work to do...

harbaum, 2025-04-09:
Does that keyboard have a different key you could re-purpose? Maybe some rarely used one? Or a long press on restore or the like? That could be done inside the FPGA Companion firmware and would not need any special support on FPGA side.

vanfanel, 2025-04-09:
> Does that keyboard have a different key you could re-purpose? Maybe some rarely used one? Or a long press on restore or the like? That could be done inside the FPGA Companion firmware and would not need any special support on FPGA side.

The keyboard doesn't have any special keys I know of, other than being a plain C64 keyboard.
But a key that has not much use on a C64 is the "pound" key, which on GNU/Linux produces scancode 0x27, according to XEV:
```
KeyPress event, serial 39, synthetic NO, window 0x1600001,
    root 0x400, subw 0x0, time 83083999, (-627,102), root:(244,550),
    state 0x10, keycode 20 (keysym 0x27, apostrophe), same_screen YES,
    XLookupString gives 1 bytes: (27) "'"
    XmbLookupString gives 1 bytes: (27) "'"
    XFilterEvent returns: False
```
I think the key should be remapped to the "pound" symbol before being used for menu toggling however?
(It has a "pound" symbol on it)

vossstef, 2025-04-09:
Hi @vanfanel ,
please consider to make use of this tiny extra Keyboard for the F12 Key topic . Some users found that helpful for A2600.
https://github.com/vossstef/A2600Nano/issues/7

harbaum, 2025-04-09:
Ah ... yes, of course.

And  one could actually make a little firmware for a pico (or pico zero) that implements a hid keyboard with just F12, the cursor keys and the return key as a 'mistery menu controller'. This would be super cheap but also flexible.

vanfanel, 2025-04-10:
Thanks for thinking this alongside me, guys!
But the solution may be a tad easier. 
On GNU/Linux, as I said, the "pund" key sends `keycode 20 (keysym 0x27, apostrophe)` right now.
Ok, I just have to remap that to F12, and presto! It has to work like that!
So.. where is the keyboard mapping file? I guess that's part of the companion programming, right?

vossstef, 2025-04-10:
@vanfanel ,
please have a look to hid.c and search for 
// F12 toggles the OSD state. Therefore F12 must never be forwarded
to get an idea

Maybe you compile for your own purposes a modified Version and try it.

vanfanel, 2025-04-11:
> [@vanfanel](https://github.com/vanfanel) , please have a look to hid.c and search for // F12 toggles the OSD state. Therefore F12 must never be forwarded to get an idea
> 
> Maybe you compile for your own purposes a modified Version and try it.

Ah, yes, that's promising. I see this line:
https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L131
...where it seems that the menu is toggled if the value in the buffer position is `0x45`.
I understand that `0x45` must be the scancode for F12, right?
If I connect the TheC64 keyboard to a GNU/Linux system, the "showkey" command tells me that the "pound" key on the keyboard has scancode 0x0c.

However, after building the latest FPGA-companion core from latest GIT code, it doesn't work with the latest stable tang_nano_20k_c64 release (1.9.1) to complete this experiment.

Can you please point me to the exact commit in https://github.com/harbaum/FPGA-Companion/commits/main/ that corresponds to the FPGA-Companion release 1.2?

vanfanel, 2025-04-11:
@vossstef Well, latest commit from March 3 doesn't build:

```
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c:18:103: error: 'CONFIG_USBHOST_ASIX_ETH_MAX_TX_SIZE' undeclared here (not in a function); did you mean 'CONFIG_USBHOST_PSC_STACKSIZE'?
   18 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_asix_rx_buffer[CONFIG_USBHOST_ASIX_ETH_MAX_TX_SIZE];
      |                                                                                                       ^~~~~~                             
      |                                                                                                       CONFIG_USBHOST_PSC_STACKSIZE
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c:15:106: error: 'CONFIG_USBHOST_RTL8152_ETH_MAX_RX_SIZE' undeclared here (not in a function)
   15 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_rtl8152_rx_buffer[CONFIG_USBHOST_RTL8152_ETH_MAX_RX_SIZE];
      |                                                                                                          ^~~~~~~~~                             
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c:19:103: error: 'CONFIG_USBHOST_ASIX_ETH_MAX_RX_SIZE' undeclared here (not in a function); did you mean 'CONFIG_USBHOST_PSC_STACKSIZE'?
   19 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_asix_tx_buffer[CONFIG_USBHOST_ASIX_ETH_MAX_RX_SIZE];
      |                                                                                                       ^~~~~~                             
      |                                                                                                       CONFIG_USBHOST_PSC_STACKSIZE
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c:16:106: error: 'CONFIG_USBHOST_RTL8152_ETH_MAX_TX_SIZE' undeclared here (not in a function)
   16 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_rtl8152_tx_buffer[CONFIG_USBHOST_RTL8152_ETH_MAX_TX_SIZE];
      |                                                                                                          ^~~~~~~~~                             
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c: In function 'usbh_asix_get_eth_txbuf':
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c:765:1: warning: control reaches end of non-void function [-Wreturn-type]
  765 | }
      | ^
At top level:
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c:19:86: warning: 'g_asix_tx_buffer' defined but not used [-Wunused-variable]
   19 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_asix_tx_buffer[CONFIG_USBHOST_ASIX_ETH_MAX_RX_SIZE];
      |                                                                                      ^~~~~~~~~~~~~~~~
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_asix.c:18:86: warning: 'g_asix_rx_buffer' defined but not used [-Wunused-variable]
   18 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_asix_rx_buffer[CONFIG_USBHOST_ASIX_ETH_MAX_TX_SIZE];
      |                                                                                      ^~~~~~~~~~~~~~~~
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c: In function 'usbh_rtl8152_get_eth_txbuf':
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c:2232:1: warning: control reaches end of non-void function [-Wreturn-type]
 2232 | }
      | ^
make[3]: *** [build_out/components/usb/cherryusb/CMakeFiles/cherryusb.dir/build.make:202: build_out/components/usb/cherryusb/CMakeFiles/cherryusb.dir/class/vendor/net/usbh_asix.c.o] Error 1
make[3]: *** Waiting for unfinished jobs....
At top level:
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c:16:86: warning: 'g_rtl8152_tx_buffer' defined but not used [-Wunused-variable]
   16 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_rtl8152_tx_buffer[CONFIG_USBHOST_RTL8152_ETH_MAX_TX_SIZE];
      |                                                                                      ^~~~~~~~~~~~~~~~~~~
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/class/vendor/net/usbh_rtl8152.c:15:86: warning: 'g_rtl8152_rx_buffer' defined but not used [-Wunused-variable]
   15 | static USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX uint8_t g_rtl8152_rx_buffer[CONFIG_USBHOST_RTL8152_ETH_MAX_RX_SIZE];
      |                                                                                      ^~~~~~~~~~~~~~~~~~~
[ 51%] Building C object CMakeFiles/app.dir/home/manuel/src/bouffalo/FPGA-Companion/src/u8g2/csrc/u8x8_cad.c.o
make[3]: *** [build_out/components/usb/cherryusb/CMakeFiles/cherryusb.dir/build.make:216: build_out/components/usb/cherryusb/CMakeFiles/cherryusb.dir/class/vendor/net/usbh_rtl8152.c.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:727: build_out/components/usb/cherryusb/CMakeFiles/cherryusb.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 51%] Building C object CMakeFiles/app.dir/home/manuel/src/bouffalo/FPGA-Companion/src/u8g2/csrc/u8x8_capture.c.o
[ 51%] Building C object CMakeFiles/app.dir/home/manuel/src/bouffalo/FPGA-Companion/src/u8g2/csrc/u8x8_d_a2printer.c.o
[... cut ...]
/home/manuel/src/bouffalo/FPGA-Companion/src/bl616/mcu_hw.c:459:3: error: too few arguments to function 'usbh_initialize'
  459 |   usbh_initialize();
      |   ^~~~~~~~~~~~~~~
In file included from /home/manuel/src/bouffalo/FPGA-Companion/src/bl616/mcu_hw.c:8:
/home/manuel/src/bouffalo/bouffalo_sdk/components/usb/cherryusb/core/usbh_core.h:276:5: note: declared here
  276 | int usbh_initialize(uint8_t busid, uintptr_t reg_base);
      |     ^~~~~~~~~~~~~~~

```

Can you please point me to a GIT code commit  that builds and works with latest stable tang_nano_20k_c64 release (1.9.1)?

vossstef, 2025-04-11:
@vanfanel , i am doing very big changes to both SDK and FPGA-companion bl616 section at the moment to add WIFI functions and i tested now almost daily againt an unpublised SDK locally stored.

Hopefully the build instruction over here: https://github.com/harbaum/FPGA-Companion/tree/main/src/bl616 are good enough to understand what to do.
You need to use again harbaums SDK repo location if you cloned some weeks ago. 

My latest code uses usbh_hid_get_report_descriptor(usb->hid_info[i].class, report_desc[i], 128); and that means native Cherry 1.4.3 is needed. Your SDK need to be latest version out of the harbaum repo location as it includes additions i did for latest native USB stack.

Following SDK's will not work anymore ! 

https://github.com/CherryUSB/bouffalo_sdk
https://github.com/vossstef/bouffalo_sdk.git
https://github.com/bouffalolab/bouffalo_sdk

only this one:
https://github.com/harbaum/bouffalo_sdk

vanfanel, 2025-04-11:
@vossstef I am using:
https://github.com/bouffalolab/toolchain_gcc_t-head_linux.git
https://github.com/harbaum/bouffalo_sdk.git
to build:
https://github.com/harbaum/FPGA-Companion.git

Latest GIT code in https://github.com/harbaum/FPGA-Companion.git DOES build correctly, but does not work with the latest core in https://github.com/vossstef/tang_nano_20k_c64/releases

What I need is a version of https://github.com/harbaum/FPGA-Companion.git that works with the latest release in https://github.com/vossstef/tang_nano_20k_c64/releases

vossstef, 2025-04-11:
@vanfanel, 
lately i included in my C64 the XML settings, keyboard mapping and a 6551 UART plus changes in hid and sysctrl to align with Till's ST and WIFI modem interface needs. 
I think you need to compile the VHDL/Verilog and flash your TN20k.  
Exactly that setting i am using also at the moment. 
Will test later on that the stored bl616 code compiles and work with https://github.com/harbaum/bouffalo_sdk as mentioned i am using even newer SDK

vanfanel, 2025-04-11:
@vossstef When you have time (I know you are busy with all those improvements!) I could use a version of the core that goes with the current Companion firmware, so please upload the core build you use and I will use that for now.

harbaum, 2025-04-11:
I am considering to add a setting to the ini file stored on SD card which will allow to change the menu hotkey on a per core basis. Something similar was needed for the original mist when the acorn Archimedes core came which actually has a real F12 key and needs that itself.

I am currently porting misterynano to the efinix Trion T20 fpga and do the menu key config and a few other small things after that.

vossstef, 2025-04-11:
> @vossstef When you have time (I know you are busy with all those improvements!) I could use a version of the core that goes with the current Companion firmware, so please upload the core build you use and I will use that for now.
> 

@vanfanel what Build env do you use? 
Windows or Linux ?

vossstef, 2025-04-11:
@vanfanel, 
i just did a build on fedora 41 and build is working like a charm.
SDK from Till's repo and of course the FPGA-companion. No build failures.
Please check the details if your build fails. Of course the bl616 binary WIFI is not functional with that SDK but no problem for well known "normal" old functions.

vanfanel, 2025-04-11:
@vossstef,
as I said, the build is OK, but the latest GIT code for the Companion doesn't work with the latest core release.
I think there is some idiomatic barrier here... :)

vossstef, 2025-04-11:
@vanfanel ,
i did FPGA companion build and C64 core build under Linux Fedora 41 and flashed both µC and TN20k. 
The core is working fine and no issues apart from known buggy WIFI.

vanfanel, 2025-04-11:
@vossstef Ok! Can you please upload the core build that you have working with current Companion GIT sources?

vossstef, 2025-04-12:
@vanfanel 
i dropped my build over here (Fedora 41)

https://github.com/harbaum/FPGA-Companion/releases/tag/v1.2.1

and the C64 core
https://github.com/vossstef/tang_nano_20k_c64/releases/tag/v1.9.2

Did only a very brief check and USB hub, xbox360 controller, the RII key combo and WIFI working fine.

vanfanel, 2025-04-12:
@vossstef Thanks!! That build of the core works with current Companion code!

What I did was changing 0x

vanfanel, 2025-04-12:
@vossstef 

Thanks, that version of the core does work with the current companion code!

I changed 0x45 in https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L131 for 0x53 which corresponds to the scancode of the "TheC64" labeled key on the keyboard, and also tried 0x0c which is the scancode for the "pound" key, then rebuilt the companion and re-flashed it. 
However, none of these scancodes work for activating the menu...
Which leads me to ask: what kind of scancodes are you using in https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L131, please?
I find the scancodes by running `showkey --scancodes` on GNU/Linux, but maybe that's not the intended method...

vossstef, 2025-04-12:
> [@vossstef](https://github.com/vossstef)
>
> Thanks, that version of the core does work with the current companion code!
>
> I changed 0x45 in [...]

@harbaum can you pls. advise about the Scancodes and how to replace the F12 by 'Pound' code

harbaum, 2025-04-12:
These USB HID codes like these:

https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

But I cannot tell you what those special keys deliver. Which key e.g. detects your PC if you press the C64 pound key?

vanfanel, 2025-04-12:
> These USB HID codes like these:
> 
> https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2
> 
> But I cannot tell you what those special keys deliver. Which key e.g. detects your PC if you press the C64 pound key?

When connected to a GNU/Linux desktop PC, the pound key produces the single quote character (this one --> ' ) and according to `showkeys --scancodes` it produces the `0x0c` scancode.

In the file you pointed (thanks for that!) I guess it should be:
`#define KEY_APOSTROPHE 0x34 // Keyboard ' and "`

So I used 0x34 in the comparason here:
https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L131
...and now I can bring the menu using the "arrow up" key  (the key that's just left to "Restore") on the TheC64 keyboard.

On the other hand, when connected to the TangNano+Companion, the pound key produces "-" (minus) on the C64 screen.
According to https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2, that's 0x2d.
So I put 0x2d in https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L131, and now I have the menu working on the TheC64 keyboard on the TangNano20K C64 core!! :)

However, an small issue remains: some keys in the TheC64 keyboard don't correspond to the symbols they produce on screen. Most keys are right, but there are like 5 that are wrong.
For example, "*" on the keyboard prints ":" on screen.
How can I adjust that on the companion code or the core's code, please?

vanfanel, 2025-04-12:
Ah, I should do the adjustments around here, it seems:
https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/core_c64.c#L162
The problem is knowing how to adapt that to the TheC64 USB keyboard... Any ideas are welcome.

vanfanel, 2025-04-12:
It seems that the keymaps in https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/core_c64.c#L162 are ununsed. Apparently they were used for "legacy cores" but not anymore.
Then I don't know where to modify the keymaps.

vossstef, 2025-04-12:
@vanfanel, 
the keyboard mapping moved lately into the FPGA core and can be found here:

https://github.com/vossstef/tang_nano_20k_c64/blob/main/src/misc/c64_keymap.v

I tried to replicate the Keyboard mapping on my German Keyboard ([RII Mini Keyboard i8](http://www.riitek.com/product/220.html)) like (MISTer):
https://github.com/vossstef/tang_nano_20k_c64/blob/main/.assets/keymap.gif
not sure how that feels on other language Keyboards..

vanfanel, 2025-04-13:
@vossstef Thanks for pointing me to the right place, it seems the keyboard mapping has been traveling back and forth from core to companion and back to core! ✈ 

Now, looking at https://github.com/vossstef/tang_nano_20k_c64/blob/c2c524513433d8a0d72023f8ad7484ba1b012dda/src/misc/c64_keymap.v#L119
...since I am getting ":" in "*", I should change:
`(code == 7'h55)?{ 3'd7,3'd1}: // 55: KP *`
...to this instead:
(code == 7'h55)?{ 3'd5,3'd5}: // 55: KP *`
...But that doesn't work because physical "*" becomes backspace... What a puzzle!

vossstef, 2025-04-13:
@vanfanel sorry but you need to try. Keyboard mapping for C64 / VIC20 was if i recall right horrible and took endless time. Apart from the fact that in VHDL-2008 the Verilog structure was not possible and i had to circumvent the interfaces.

vanfanel, 2025-04-14:
> [@vanfanel](https://github.com/vanfanel) sorry but you need to try. Keyboard mapping for C64 / VIC20 was if i recall right horrible and took endless time. Apart from the fact that in VHDL-2008 the Verilog structure was not possible and i had to circumvent the interfaces.

Ok, thanks! I am slowly building a keymap for the TheC64 keyboard. When its ready I will post It here :)

vanfanel, 2025-04-15:
@vossstef I'm almost there with the correct mapping! I am only missing the RESTORE key code to send.
The Readme says that RESTORE is mapped in F11, but looking at the F11 mapping in https://github.com/vossstef/tang_nano_20k_c64/blob/c2c524513433d8a0d72023f8ad7484ba1b012dda/src/misc/c64_keymap.v#L96, I see it's sending `3'd7,3'd1`  which is the code for LSHIFT.

What's the code I should send for RESTORE?
I know it's not on the C64 matrix, according to https://www.c64-wiki.com/wiki/Keyboard,
so maybe it's being treated as a special case in the companion firmware just like F12, but.. where?
It's mentioned in https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L52 but nothing else...

vossstef, 2025-04-15:
Restore is used for triggering nmi interrupt in c64

vanfanel, 2025-04-15:
> Restore is used for triggering nmi interrupt in c64

Yes, I know. But where is it mapped? In the core? Or in the Companion?
Can you point me to where it's mapped, please?

vanfanel, 2025-04-15:
> [@vanfanel](https://github.com/vanfanel), basically i am using a 2nd communication channel from companion originally intended for Joystick Numpad... numpad(6) for NMI (F11) and numpad(7) for swap of joystick ports. I think we have at the moment a hard mapping for those Keyboard joystick still in companion.

Can you point me to the code of that hard mapping, please?

vossstef, 2025-04-15:
@vanfanel ,
it's in the hid.c line 39 and following.

vanfanel, 2025-04-16:
> [@vanfanel](https://github.com/vanfanel) , it's in the hid.c line 39 and following.

This exact line is where F11 is assigned to RESTORE:
https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L66
So setting that to 0x2b instead of 0x44, I got the physical RESTORE key working on the TheC64 keyboard!

vanfanel, 2025-04-16:
@vossstef Now, RESTORE (0x2b) seems to activate both RESTORE and LSHIFT, as in:

https://github.com/user-attachments/assets/7ac33207-11b2-45ef-922a-ec0c240cfcc2

So, to prevent that, I changed this:
https://github.com/harbaum/FPGA-Companion/blob/dc5af8d6a0e69a1a1df5227406d4f1e999a6da47/src/hid.c#L134
For this:
`if(!osd_is_visible() && (buffer[2+i] != 0x2b)) /* Don't send the RESTORE key to the core */`
...so the key isn't sent to the core as a normal key.

...and now, RESTORE is working as intended! Yeah!

https://github.com/user-attachments/assets/c17efb11-40af-4f35-850d-caad0e030051

vossstef, 2025-04-16:
@vanfanel, 
congratulations  and great success !

have not much time for core at the moment but think as keymapping is inside the core it should be easier possible to extract Numpad and those extra handlings like restore.  Need to dig into that...

vanfanel, 2025-04-17:
@vossstef Another small question if you don't mind.
In this line:
https://github.com/harbaum/FPGA-Companion/blob/755c88de03dc643b59d2de5fe4c6c9cf6170dabf/src/hid.c#L139
you guys detect the UP arrow keypress inside the GUI.
However, the TheC64 doesn't have an UP key, it has the same DOWN/UP key as a C64 keyboard, with UP needing SHIFT+DOWN/UP.
So, is there a way for me to detect that SHIFT is pressed?

vossstef, 2025-04-17:
@vanfanel , same idea i had also more than a year ago but didn' found time to think about.

vanfanel, 2025-04-17:
@vossstef Ok, no hurries, thanks for the answer!

What about SHIFT LOCK? Using a normal USB keyboard, the CAPS LOCK key doesn't work as it does on a real C64: it acts as LSHIFT. 
On a real C64, SHIFT LOCK lets the user change lettercase permanently.

harbaum, 2025-04-22:
I just commited a change to the FPGA Companion that will allow you to specify the key in the ini file:

```
; firmware options
; HID key code of OSD/menu hotkey
option hotkey=69

```

vossstef, 2025-04-22:
> [@vossstef](https://github.com/vossstef) Ok, no hurries, thanks for the answer!
> 
> What about SHIFT LOCK? Using a normal USB keyboard, the CAPS LOCK key doesn't work as it does on a real C64: it acts as LSHIFT. On a real C64, SHIFT LOCK lets the user change lettercase permanently. It's not a normal key: it's a physical electric latch. That doesn't seem to be the case on the core.

@vanfanel i need to adjust the c64 / vic20 core a bit

vossstef, 2025-04-23:
@vanfanel let me first update the core to cover the shift cursor and shift Fx Key topc and will look then into the RESTORE too.

vossstef, 2025-04-29:
@vanfanel 
i did't forgot you... Just created a pre-release for you that already adds the auto shift for Cursor up/left and the Function Keys.
https://github.com/vossstef/tang_nano_20k_c64/releases/tag/v2.0.1

CAPS Lock need also a fix and still need to do a refactoring in the Numpad area.