MiSTle-Dev/C64Nano #137: Misteryshield20k_rpipico - keyboard/mouse USB doesn't work D2,D3 is on

luciust, 2025-04-24:
The setup: soldered the board, powering via Tang Nano 20k USB-C, the power draw is 1W just as if the Tang is running standalone. Atari ST core starts up and I get the desktop - I'm using original working Atari ST core here before moving to C64.

So using the https://github.com/vossstef/tang_nano_20k_c64/blob/main/board/misteryshield20k_rpipico/README.md board, and found out so far:
- the FPGA Companion 1.2 release (fpga_companion.uf2) works with RPI Pico 1 (original board)
- the FPGA Companion 1.2 release (fpga_companion_rp2040_zero.uf2) works with RPI Pico W

by works I mean I get D2 LED on pretty much at boot and couple seconds later D3 also lights up. D4 is silent, and it stays this way - nothing is connected.

The Problem: When connecting mouse, keyboard to USB - all I get is brief flash of the IR LED in mouse and then it's off. **Neither work.** D2-D4 stay the same, Tang 20K works during all the USB connections. 

My suspision is that the 12MHz crystal might be wrong or soldered incorrectly, other stuff on the board should be OK. Any pointers what I should look at now ? 
I do have the chinese RPi Pico (those in purple with 16MB Flash) which are good for making Atari XL cartiges...

harbaum, 2025-04-24:
There's a lot I don't understand in your report. The pi zero does not fit onto the shield and the version for the waveshare zero also won't work on the regular pi pico as the USB is wired up differently.

luciust, 2025-04-24:
I'm using RPi Pico W and RPi Pico here, not the "pi zero". Actually moved to RPi Pico since the I thought the fpga_companion.uf2 was for RPi Pico W - but it didn't work - there was no LED after copying the uf2 onto it - but the RPi Pico 1 - the first original board did start with it.

As for the "whaveshare zero" - I do have this one: https://www.aliexpress.com/item/1005008250289610.html but I haven't  tested yet - need to make sure it's pin compatible (it's suppose to be)

luciust, 2025-04-24:
It seems I have similar situation: https://github.com/harbaum/FPGA-Companion/issues/36

I've copied over u2f on original Pico (not Pico W) and 
- it flashes it's green LED constatnly when connected to the mysteryshiled29k_rpipico, but D2,D3,D4 does not light up
- it flahsed 13 times when connected as standalone - to power only

Here's the video when RPi Pico is conneced to the shield: https://www.youtube.com/watch?v=4EKl9PAC8ww

Investingating further - this might be related to copying file vs. flashing...

luciust, 2025-04-24:
Further investigation on Pico's:

I've build the fpga_companion.uf2 from the source and copied to RPi Pico and RPi Pico W:
What changed: 
- both Pico and Pico flash it's green LED when connected to shield
- D2,D3 and D4 are off 
- mouse, keyboard - don't work they get a flash of their LED but no furhter action
- Tang Nano 20K doesn't boot the ST core properly - it stays at the white screen with two orange LED on and crashes there - no more reaction to the reset button. When Pico is disconnected - Tang boots Atari ST to the desktop.

Internesting!

luciust, 2025-04-24:
I got something!

```
SYS: Core ID: 01
SYS: Core: Atari ST
SYS: FPGA ready after 3070ms!
SYS: SYS set value R = 3
SDC: ---- SDC init ----
SDC: SD not ready, status = 34
config[0] = 01
No valid config found, neither on sd card nor in core
```

harbaum, 2025-04-24:
So what are you missing? No config found just means that it's a rather old core and uses the old method. But that's no error and should just work.

You don't get any output regarding detected USB devices? If you have some USB peripherals connected you should see some debug output regarding USB.

luciust, 2025-04-24:
Yes, got old core (previous), and I'm missing the USB. So crystal and CH334U are the suspects, right?

luciust, 2025-04-24:
Just looking at the schematics - I think it's gonna be the crystal - I need to check it with scope...

[image]

luciust, 2025-04-24:
[image]

luciust, 2025-04-26:
Well it *WAS* the 12MHz crystal.

Let's recap:

- I've soldered the board with wrong crystal - it was not the PIN 1 XI, 2 GND, 3 XO, 4 GND but VCC, GND, OUTPUT. I got the right crystal 12MHz:

[image]

- but the problem was the orientation, first of all the denoted PIN 1 (supposed XI) was shorted with PIN 3 (supposed XO) - so it seemed that there's a wrong description. Soldered it anyway - and it still did not work
- based on the findings I've moved it 45 deg because the XI and XO cannot be shorted and... it worked! USB hub is working fine. 
Here's the final position of my crytstal, I'll order more and check in 3-4 weeks when they arrive:

[image]

closed 2025-04-26 by luciust

harbaum, 2025-04-26:
Ok ... wasn't aware that you soldered the board yourself. I thought you got it fully assembled and I thus wasn't expecting the crystal to be the problem.

If you like to improve your soldering skills: The next variant may have the rp2040 directly on the PCB, hopefully underneath the Tang Nano 20k and thus allowing the board to be smaller. I like small devices ...

harbaum, 2025-04-26:
BTW: You have a Pico-W, so you should now be able to go online with your setup

luciust, 2025-04-26:
I saw the updated release, and I'm gonna test the online ASAP :)

Well, here I was able to resolder the crystal 6 times - so it was great for getting the skill of fixing SMD in tight spacer. Thanks!

vossstef, 2025-04-27:
have a look in the c64 readme how to usevthe wifi modem.  I tried to document the steps.

luciust, 2025-04-28:
> have a look in the c64 readme how to usevthe wifi modem. I tried to document the steps.

Yay, it works! Steps were clear athelp did the job. I'm ONLINE for the first time using Atari ST!

[image]